GeekMagicClock/smalltv-ultra #133: ip port connect problem

dmkn96, 2025-03-18:
Hello, thank you for supporting this product, I really like it.
After updating to version 9.0.31, the device connects to my Wi-Fi and launches without any problems, 
but after completing the setup, I can't open the controller page using the IP address shown at the beginning.

[image]

GeekMagicClock, 2025-03-18:
Hi, please check the IP address displayed on the screen, it may be changed by the router, and make sure your PC and the clock are under the same local network.

dmkn96, 2025-03-18:
Yes, I did exactly that, but I'm not sure why the system isn’t working correctly after the update. I can't connect to the homepage, even though the device is 100% connected to my Wi-Fi (all the weather data appears). When I turned on the device in an area without Wi-Fi, I was finally able to connect to the non-connected homepage at 192.168.4.1. However, there were a lot of artifacts, the text didn’t load properly, and there were no options in the dropdown fields. I downgraded the update to 9.0.26 in the hopes that it would help. Could you confirm whether version 9.0.31 is stable?

dmkn96, 2025-03-18:
[image]
Damn, it looks like my system is broken after the 9.0.31 update... nothing is working correctly.

dmkn96, 2025-03-18:
wops im idiot i guess i just updated wrong file

closed 2025-03-18 by dmkn96