GeekMagicClock/smalltv-ultra #135: forecast Issue

Flipsanddips, 2025-03-31:
Says 0% in red does not change. Other side input looks static maybe It changes.

Flipsanddips, 2025-03-31:
O% 32/32

cdelavega950, 2025-03-31:
tomorrow it will work

GeekMagicClock, 2025-04-02:
@Flipsanddips  Please update to V9.0.31 and follow the instructions to set your own key.

Flipsanddips, 2025-04-02:
Thank you lad <3

Flipsanddips, 2025-04-02:
Thank you very much!

Flipsanddips, 2025-04-02:
it works!

Flipsanddips, 2025-04-02:
I also have a Pro model; it constantly reboots. Any ideas?

butinhi, 2025-04-22:
How to set this temp
[url=https://ibb.co/DgfKvH55][img]https://i.ibb.co/DgfKvH55/photo-2025-04-22-10-12-33.jpg[/img][/url]

GeekMagicClock, 2025-04-22:
> How to set this temp [image]

Did you use your own forecast api key?

butinhi, 2025-04-22:
> > How to set this temp [image]
> 
> Did you use your own forecast api key?

Yes, I have

GeekMagicClock, 2025-04-23:
[image] That should be the avarage temp, you  can try.

butinhi, 2025-04-23:
> [image] That should be the avarage temp, you can try.

Working. Thanks!

closed 2025-05-09 by GeekMagicClock