GeekMagicClock/smalltv-ultra #138: Idea

Olxsio, 2025-05-01:
PLEASE add a feature that will display your phone's notifications on the device.
Maybe it would require an app to be installed on the smartphone.
*pleeaaase*

Razbat4444, 2025-05-02:
i agree that would be an amazing feature 
can use blutooth to transmit and receive notification data

mnemologic, 2025-05-08:
> i agree that would be an amazing feature can use blutooth to transmit and receive notification data

ESP-12E module inside SmallTV-Ultra does not have Bluetooth implemented.
So hardware incompatibility. Maybe future versions will embrace this idea.

GeekMagicClock, 2025-05-09:
Hello, what info would you need to notify to your smartphone from the clock?
There're some third apps can receive notification via internet.

Olxsio, 2025-05-13:
> Hello, what info would you need to notify to your smartphone from the clock? There're some third apps can receive notification via internet.

Basically the notifications. I'd like to see the most recent notification on the screen of the device. *please*