GeekMagicClock/smalltv-ultra #139: City Name Bug (Inconsistent Display)

SirChokealot, 2025-05-09:
The City name in the weather page does not display consistently.

It does not always happen when turning on the device, sometimes it's correct, sometimes it's wrong.

It's a short name and can display fully at times but it's inconsistent and sometimes displays Jersey Ci instead of Jersey City

[image]
[image]

It can easily be fixed with a reboot of the device.

GeekMagicClock, 2025-05-09:
OK, thanks for report, will check and fix this.

GeekMagicClock, 2025-05-09:
Hi, I tried to repreduce this problem with your city number and several reboot, but just no luck.

SirChokealot, 2025-05-09:
If it matters, I usually encounter this bug from an unplugged state (I unplug the power strip of my whole set-up every night)>

I've never encountered it doing a web reboot from 192.168.1.***